jneilliii/OctoPrint-PrusaSlicerThumbnails #106: Preview stopped working

nicktodorov, 2023-03-31:
After some time working normally, preview pictures stopped loading. There are errors in browser console for missing file. 
With ssh I can see that in the folder there is a png with the same file name as the gcode - test5.png and test5.gcode. But the strange thing is that the error in console says it cannot find a file with shrange posfix after the name - "http://xxx.xxx.xxx.xxx/downloads/files/local/test5.png?20230331132745"

jneilliii, 2023-03-31:
This appears to be using the use uploads folder?

jneilliii, 2023-03-31:
What version of OctoPrint?

nicktodorov, 2023-03-31:
Sorry for not pointing it - 1.8.7

jneilliii, 2023-03-31:
weird, I can't seem to reproduce this issue. I just tested on my instance with the option to save images into uploads folder and it definitely generated the image and stored the file, as you can see in my file list screenshot below the png file shows and the button appears to open the preview (with inline thumbnails option disabled). 

[image]

nicktodorov, 2023-03-31:
Sorry for the stupid question, but can you please explain "the option to save images into uploads folder"? Maybe I am missing something?
I can also see the preview button, I can also make it try showing something with inline, the problem is that it is no picture but the question mark icon when a picture is not loaded on a webpage.

jneilliii, 2023-04-01:
This option:

[image]

nicktodorov, 2023-04-01:
Ok I have it.
But noticed that scan files button also says no images.
[image]

Here is my Cura settings:

[image]

And here the beginning of the gcode:

[image]

Please, help me figure out what I am doing wrong.

jneilliii, 2023-04-01:
That option is for a very specific use case of multiple instances sharing the same uploads folder. If you don't have that try disabling the option and uploading a new file and see if things work as expected.

nicktodorov, 2023-04-01:
Thank you so much!
Yes, after disabling it preview showed up. 
Enhancement proposal: note this specific use somehow in the settings to avoid other users struggling like me in the future?

jneilliii, 2023-04-01:
It's disabled by default, but yes, documentation does need updating. I'm still confused why it didn't work for you though in the other mode.

nicktodorov, 2023-04-01:
Fixed.
Thanks again.

closed 2023-04-01 by nicktodorov