jneilliii/OctoPrint-PrusaSlicerThumbnails #107: Cached issues with ongoing Octoprint instances

bolsoncerrado, 2023-04-16:
Cheers again,

I sort of "reopen" this issue as there must be something going on somewhere still and I'll try to explain/open to questions to further troubleshoot this.

As you may remember I use a mapped "uploads" folder in a NAS where all the Prusas feed from when printing from their local raspberries running Octoprint (one raspi per printer).

Recently the NAS flagged the drive as problematic (mechanical one) and i replaced it with a SSD so instead of copying over the old files, I just reset the folder and start clean.

RIGHT AFTER booting all the printers (and octoprints) in the farm, I started to upload the files I required for a print job. The thing is, this first upload generated a thumbnail which was VIEWABLE by all the printers in the farm!

Note the following image where the BO KATAN SILVER file  has the thumbnail associated icon viewable by several printers including the ones where the file was not "dropped" for upload:

[image]

Today after the print job of last night was completed I uploaded a new one, the ELYSIUM silver one.

As you can notice in the image above, only one of the three show the thumbnail icon no matter how many refreshes I do on the other two. Furthermore I had to upload it twice (overwritting the file) in order for the left most to detect the embeded thumbnail.

This makes no sense at all!!

Not sure if you need further clarification but the main question is:

IF IT WORKED ONE TIME (KATAN SILVER) WHY IT WOULD FAIL ON SUBSEQUENT UPLOADS?!?!?

Thanks for your efforts!

jneilliii, 2023-04-16:
> IF IT WORKED ONE TIME (KATAN SILVER) WHY IT WOULD FAIL ON SUBSEQUENT UPLOADS?!?!?

Logically there's no reason I can think of. Check browser developer tools for file not going errors to validate it doesn't see the file?

bolsoncerrado, 2023-04-16:
Not sure if I understood your message.

To check browser developer console? To find what? File gets uploaded fine. It's just the thumbnail not being parsed afterwards.

For subsequent uploads to this "no longer empty folder", I have had to upload and then overwrite the same file again in order for the thumbnail to show up, but the rest of machines no longer see the thumbnail.

jneilliii, 2023-04-16:
that's what I'm trying to figure out, it the broken instances are even seeing the png file or not. Try this instead. on a working instance, right click the image and select copy image address. then paste that into your browser's address bar, but change the beginning of the URL to the broken instance address and see if it loads.

jneilliii, 2023-04-16:
I think I might see a potential cause of this. If you don't have custom background plugin on the broken instances then the download URL for the thumbnail gets broken because png files are no longer a valid filetype for OctoPrint.

bolsoncerrado, 2023-04-16:
I do have background plug-in on all instances as they're basically the same SD image copied all over and just the IP and printer name changes....

I'll test the image path thing and report back.

bolsoncerrado, 2023-04-16:
OK tested and YES, "Broken" instance IP does see the PNG if entered directly. 

Alsoo note in the following video that:

1) Pis have been running nonstop for 48h now
2) I uploaded the lexington file on the left Pi  and no thumbnail
3) I REUPLOADED and OVERWROTE lexington file on the left Pi and the thumbnail icon appears, but NO thumbnail is loaded
4) the right Pi does not see the file thumbnail at all (no icon)
5) I enter the URL of the image of the left Pi on a new window and change the IP of the right PI and boom. Image loads fine.

https://www.dropbox.com/s/b7j5kuysld8vqh8/Grabaci%C3%B3n%20de%20pantalla%202023-04-16%20a%20las%2019.23.16.mp4?dl=0

In the meantime I'll check with the browser cache settings (currently using Vivaldi, but AFAIR it also happened with Firefox). But sounds unlikely to me it's related to that as the octoprint spinning wheel sometimes does not trigger....

jneilliii, 2023-04-16:
ok, so the file list will show the button if the metadata is picked up that contains the URL path added by the plugin. on those same broken instances, if you expand the file data do you see model size, filament, and estimate print time?

bolsoncerrado, 2023-04-16:
Expand the file data like how?
Pressing the LOAD icon u mean?

jneilliii, 2023-04-16:
the little arrow next to the file name.

bolsoncerrado, 2023-04-16:
Good catch. That arrow is grayed out and can't do nothing on the "broken" Pi but I can start the print job without issues.

bolsoncerrado, 2023-04-16:
OK what about this one:

1) choose a pi
2) upload the gcode
3) thumbnail does not show up
4) reupload+overwrite
5) the thumbnail does show up
6) little arrow grayed out even if everything works as expected (on second upload though)

[image]

Just note ALL files show the arrow grayed out, not just the one I just uploaded twice.

bolsoncerrado, 2023-04-16:
To further addo, rebooting the whole Pi won't make a difference.

bolsoncerrado, 2023-04-16:
And this is the only reference I could find on the VIVALDI settings that speaks of "cache" anything and it's NOT set to cache anyway.

[image]

bolsoncerrado, 2023-04-16:
And finally I can confirm dropdown arrow for file info is DISABLED for every and all Pis in the farm, if that helps troubleshoot.

jneilliii, 2023-04-16:
Well until that arrow isn't disabled to show additional information there's nothing much I can do because those instances aren't getting the metadata that has the thumbnail URL included. This use outside my control and Italy one of the reasons OctoPrint maintainer didn't recommend a shared uploads folder. If you force refresh the page on those other instances does anything come back?

bolsoncerrado, 2023-04-16:
There's no longer "other instances" as all show the arrow grayed out, please [see above](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/issues/107#issuecomment-1510444413)

bolsoncerrado, 2023-04-16:
But then again it worked fine ok the first 2 uploads...
Oh well....

jneilliii, 2023-04-16:
I don't understand why the other instances don't see the metadata file, as it's stored directly in the uploads folder. Once one instance has processed and updated that file (.metadata.json) it should be accessible by all others. One question that recently came up in discord that might be relevant, can you share your .service files for each of the instances?

bolsoncerrado, 2023-04-17:
I'm sorry. What files are those and where are they located. Granted all instances are now problematic.

bolsoncerrado, 2023-04-17:
OK more mystery stuff. This morning I approach the farm to harvest night printjobs.

I find out SOME instances have the Fileinfo arrow enabled after completing and some others don't.

LOL

So let me use that to try giving you the previous info you requested.

bolsoncerrado, 2023-04-17:
Well Im still puzzled. Why just a recently uploaded file won't enable the fileinfo arrow? makes no sense to me the randomness of availability of such arrow. I can't see a pattern. Let me know if you have any questions to the following videos:

https://www.dropbox.com/s/n64m5l30ukarl2i/Grabaci%C3%B3n%20de%20pantalla%202023-04-17%20a%20las%208.33.38.mp4?dl=0

https://www.dropbox.com/s/d96y92b72sko7c6/Grabaci%C3%B3n%20de%20pantalla%202023-04-17%20a%20las%208.35.20.mp4?dl=0

bolsoncerrado, 2023-04-17:
And if you zoom in this pic, you'll  notice the very first file I uploaded has the thumbnail icon present on ALL instances, but the subsequent one only shows it in ONE instance.

https://www.dropbox.com/s/alhragj2ujpgx08/Captura%20de%20pantalla%202023-04-17%20a%20las%208.45.43.png?dl=0

jneilliii, 2023-04-17:
Do you have printtimegenius plugin installed? If so, try setting the option to not wait for print completion to run analysis (not sure exact phrase/setting).

bolsoncerrado, 2023-04-17:
> I'm sorry. What files are those and where are they located. Granted all instances are now problematic.

So which files do you need?

bolsoncerrado, 2023-04-17:
> One question that recently came up in discord that might be relevant, can you share your .service files for each of the instances?

So which files do you need?

jneilliii, 2023-04-17:
> No I don't. Should I try it?

I would definitely try that plugin, it's more efficient at gcode analysis and print time estimates, and with the option to run even while printing will allow that full metadata information to become available sooner I think. 

> So which files do you need?

This is dependent on what you named the services when setting them up. If you use the octoprint_deploy script to setup your multiple instances on the same pi then those service names will match the instance name you created. The location of the files are `/etc/system.d/system/` I think, but don't currently have access to a device to confirm.

bolsoncerrado, 2023-04-17:
Oh no, each printer run its own pi. One instance per pi per printer only.

Re the plug-in i meant if it's relevant to this issue.

jneilliii, 2023-04-17:
Ah, ok. each printer on it's own pi. I do not know if it's relevant to this issue, but the reason the little arrow is disabled is because OctoPrint doesn't seem to know that the metadata file is there or updated. With a shared uploads folder I assume permissions come into play and wonder do you see the .metadata.json file from the broken instances or not via ssh?

bolsoncerrado, 2023-04-17:
Yes, all of them....

[image]

So may be some sort of semaphore re file access may explain why the very first try works but not the rest?

bolsoncerrado, 2023-04-17:
Take a look to the first lines of the metadata... I can't see anything special on the first entry whose thumbnail is seen by all of the instances and the second one that is seen only in one.

>{
  "STAR WARS BO KATAN WESTAR35 PLATE 2 SILVER.gcode": {
    "hash": "c034ee1f31d7cd1ab0f6156585defe1bdd1c0e06",
    "thumbnail": "plugin/prusaslicerthumbnails/thumbnail/STAR%20WARS%20BO%20KATAN%20WESTAR35%20PLATE%202%20SILVER.png?20230414174300",
    "thumbnail_src": "prusaslicerthumbnails"
  },
  "STAR WARS SOLO BLASTER BLACK.gcode": {
    "hash": "61b3a73c3bba4a6ea090d16927c153d6c959681f",
    "thumbnail": "plugin/prusaslicerthumbnails/thumbnail/STAR%20WARS%20SOLO%20BLASTER%20BLACK.png?20230414174443",
    "thumbnail_src": "prusaslicerthumbnails"
  },
  "INDIANA JONES M1917 REVOLVER.gcode": {
    "hash": "36fa436707b5dbfca2629bf835f9a68031e253ee",
    "thumbnail": "plugin/prusaslicerthumbnails/thumbnail/INDIANA%20JONES%20M1917%20REVOLVER.png?20230414174522",
    "thumbnail_src": "prusaslicerthumbnails"
  },
  "STAR WARS BO KATAN WESTAR35 PLATE 1 BICOLOR.gcode": {
    "hash": "3279f9a4832971dd42285fb3f398b2ec140f7531",
    "thumbnail": "plugin/prusaslicerthumbnails/thumbnail/STAR%20WARS%20BO%20KATAN%20WESTAR35%20PLATE%201%20BICOLOR.png?20230414174844",
    "thumbnail_src": "prusaslicerthumbnails",
    "history": [
      {
        "timestamp": 1681508253.1947858,
        "printTime": 20924.632452184,
        "success": true,
        "printerProfile": "_default"
      }
    ],
    "statistics": {
      "averagePrintTime": {
        "_default": 20924.632452184
      },
      "lastPrintTime": {
        "_default": 20924.632452184
      }
    },
    "analysis": {
      "printingArea": {
        "maxX": 239.048,
        "maxY": 206.277,
        "maxZ": 95.25,
        "minX": 0.0,
        "minY": -3.0,
        "minZ": 0.0
      },
      "dimensions": {
        "depth": 209.277,
        "height": 95.25,
        "width": 239.048
      },
      "estimatedPrintTime": 13279.125639910502,
      "filament": {
        "tool0": {
          "length": 41246.01349999321,
          "volume": 99.20828870423064
        }
      }
    }
  },
  "TASER X2 PARTS x 4.gcode": {},
  "ELYSIUM ASGARI PISTOL.gcode": {
    "hash": "4814cc1391ed8967130490d3cb49a9788a24b1b7",
    "thumbnail": "plugin/prusaslicerthumbnails/thumbnail/ELYSIUM%20ASGARI%20PISTOL.png?20230416104045",
    "thumbnail_src": "prusaslicerthumbnails",
    "history": [
      {
        "timestamp": 1681668644.5003617,
        "printTime": 34189.829589145025,
        "success": true,
        "printerProfile": "_default"
      }
    ],
    "statistics": {
      "averagePrintTime": {
        "_default": 34189.829589145025
      },
      "lastPrintTime": {
        "_default": 34189.829589145025
      }
    },
    "analysis": {
      "printingArea": {
        "maxX": 228.467,
        "maxY": 173.775,
        "maxZ": 134.25,
        "minX": 0.0,
        "minY": -3.0,
        "minZ": 0.0
      },
      "dimensions": {
        "depth": 176.775,
        "height": 134.25,
        "width": 228.467
      },
      "estimatedPrintTime": 19016.83575870689,
      "filament": {
        "tool0": {
          "length": 65798.47369996336,
          "volume": 158.26387621981374
        }
      }
    }

bolsoncerrado, 2023-04-17:
Also IDK why only the last 2 have the whole file info but not the rest... THat may have to do with the "shared" uploads folder issues, **but it still does NOT explain why the very first one works on all instances!**

jneilliii, 2023-04-17:
ok, thanks the actual metadata helps. looking at the URLs it has added the option for save files to upload folder is not set somewhere because the URL is not going to the downloads url but the plugin's URL `plugin/prusaslicerthumbnails/thumbnail/...` I think.

jneilliii, 2023-04-17:
The only way that would work is if you had a symlink for the /.octoprint/data/prusalicerthumbnails/ to point to a single shared location.

bolsoncerrado, 2023-04-17:
So the mapping method I use is adding these lines to the /etc/fstab file:

//192.168.1.192/Public/octoprint_uploads /home/pi/.octoprint/uploads cifs guest,nosetuids,noperm 0 0
//192.168.1.192/Public/octoprint_thumbs /home/pi/.octoprint/data/prusaslicerthumbnails/ cifs guest,nosetuids,noperm 0 0

and that's all. 

Based on your last 2 messages, what dfo you suggest me to try?

I don't know wdy mean about the URLs not going to the downloads URL....

Isn't fstab enough instead of the symlink you mention? Or should I force a symlink anyway?

I also found this alternate method of mapping a NAS for Octoprint, not sure if the way he sets it up rings any bell to you re being more optiomal than the fstab method?

https://www.youtube.com/watch?v=TgySREwiP8w

jneilliii, 2023-04-17:
ok, this is beyond my knowledge, but your approach does appear to be sharing the same central location for the plugin's data folder, which would mean you don't need to use the store images in upload folder. I can't really say one way or another why this doesn't work anymore for you. Nothing was changed relative to how things are extracted and stored, or linked to.

bolsoncerrado, 2023-04-17:
Hmmm ok but what about the location you mentioned being wrongly stored?

jneilliii, 2023-04-17:
With the shared plugin data folder that comment is no longer valid.

jneilliii, 2023-07-08:
I've re-worked how the plugin utilizes the uploads folder for shared instances like your situation. I suspect this will have the same issue as the fstab to the plugin's data folder, but means you can simplify your fstab and only have the one for the uploads folder and enable that option in the plugin's settings to save to the centralized uploads folder. I would be interested in knowing if it has any effect in this situation.

bolsoncerrado, 2023-07-08:
Will try and report back. Many thanks.

bolsoncerrado, 2023-07-08:
So far RC1 progress bar looks gorgeous ;) Thanks again ;)

But re the rework of the plugin, IDK what to tell you, things I've instantly noticed are the thumbnail processing takes longer to process and the failure rate has increased. I've tried multiupload (to 3 different instances at the same time, 3 different files) and to a single instance with no network activity. In the first case 2 of the 3 uploads didn't show the thumbnail and I had to reupload it. On the second case, no thumbnail, I had to reupload.

So for me, 1.2 was working way better and had less error rate :?

bolsoncerrado, 2023-07-08:
Hmmm and now Im having a bunch of succesful and quick uploads. Perhaps the network flucked on the previous post idk.

I'll keep testing a bit more and report back once again.

bolsoncerrado, 2023-07-09:
Seems I have a faulty RJ45 plug and sometimes the link I send files from go from 1000Mbps to 100Mbps... That might have explained slowness on teh first tests.

Will keep testing more.

jneilliii, 2023-09-28:
just following up on this issue. I just released 1.0.3 as stable and curious if you're still experiencing this issue between your instances and shared uploads folder?

bolsoncerrado, 2023-09-28:
I now just get very ocasional lack of thumbnails that seem to autpresolve by normal use of Octoprint (refresh, etc.) so it works very well for my purposes now.

Thanks!

closed 2023-09-28 by jneilliii