jneilliii/OctoPrint-PrusaSlicerThumbnails #111: Here we go again....with a REQUEST

bolsoncerrado, 2023-05-16:
I'm not sure of the streamline of the plugin vs the upload process but here's my request: a small icon next to the thumbnail to "FORCE EXTRACT" thumbnail on a given filename if it was already there...

I still find VERY ODD that when the NAS drive I'm filling the octoprint uploads and thumbnails (different folders) shows the thumbnail of the very first upload on ALL octoprint instances of our farm but next ones do not.

So perhaps a "force extract" button (or "force process") would fix it for the offending instances :?

Doable?  If not on the main branch perhaps a fork?

bolsoncerrado, 2023-05-16:
Ps. You may want to edit the Simplify3D section in the [main page](https://plugins.octoprint.org/plugins/prusaslicerthumbnails/) with v5.1 native support for thumbnails ;)

[image]

jneilliii, 2023-05-18:
> Ps. You may want to edit the Simplify3D section in the [main page](https://plugins.octoprint.org/plugins/prusaslicerthumbnails/) with v5.1 native support for thumbnails ;)
> 
> [image]

Thanks for this info, readme updated. 

As for your request, I don't see a reason why I could add that. There is the crawl approach in the settings already, and curious if you run that on a broken instance where the thumbnails do not show what happens.

bolsoncerrado, 2023-05-19:
Ah u mean the Rescan all files when pressing refresh in file list? I didn't even notice it!! Lemme try.

bolsoncerrado, 2023-05-19:
Doh! Guess what? It worked!!! Thanks again!!!

bolsoncerrado, 2023-05-19:
Now that I think of it.... Given it works by rescanning (manually with the scan button at least) the files in each instance, can you think of any trick to automate this? Cron job? Extra call after upload takes place? It would ease things on a Nas farm, really :)

bolsoncerrado, 2023-05-19:
Oh and just for the record, having enabled the Rescan upon refresh didn't work for me.

bolsoncerrado, 2023-05-19:
Yeah I forgot to click save  the setting, that's why I deleted it, it works fine. I'm only worried if the uploads folder starts to grow and each rescan takes ages I won't be able to start a print locally Hmmm well we will see based on field experience, thanks again :)

closed 2023-05-19 by bolsoncerrado

jneilliii, 2023-05-19:
For completeness, if you wanted to go the route of forcing this through cron or event manager (OctoPrint settings). The command would require an API key. In theory the following would spawn the same crawl process. Replace `YOUR_API_KEY` and `OCTOPRINT_INSTANCE_URL` with the appropriate values. Don't use the global API key, create a new Application key instead or use a user API key from user settings. 

```
curl -s -H "Content-Type: application/json" -H "X-Api-Key: YOUR_API_KEY" -X POST -d '{"command": "crawl_files"}' http://OCTOPRINT_INSTANCE_URL/api/plugin/prusaslicerthumbnails
```