jneilliii/OctoPrint-PrusaSlicerThumbnails #114: Prusa MINIs new firmware is using QOI image format instead of PNG

dmyers7, 2023-09-24:
In the new Prusa Mini Alpha firmware, implementing input shaping, Prusa has switched to a QOI image format.  Can the plugin be update to detect the format and continue showing the thumbnails?  Does the pixel count matter?

jneilliii, 2023-09-24:
I'm not familiar with that image format, but assuming pillow can read it I don't see any reason changes couldn't be implemented. Would need to know if they are flagging it differently somehow or not, can you share a sliced file?

jneilliii, 2023-09-24:
Appears to be available in Pillow version 9.5.0+. https://pillow.readthedocs.io/en/stable/handbook/image-file-formats.html#qoi, currently no restriction on Pillow version required by the plugin, so would need to add that minimum requirement in setup.py.

dmyers7, 2023-09-27:
Yup, looks like they do call it out:
"; thumbnail_QOI begin 200x240 13184"

File attached
[Sensor Body V5_0.4n_0.2mm_PLA_MINIIS_14m.gcode.zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/12732687/Sensor.Body.V5_0.4n_0.2mm_PLA_MINIIS_14m.gcode.zip)

jneilliii, 2023-09-27:
added in the above commit and available in version 1.0.3rc3. You can change your release channel in OctoPrint's software update settings for the plugin to `Release Candidate` and upgrade when prompted to test it out and validate. 

[image]

dmyers7, 2023-09-28:
Verified and validated.  It recovered images in QOI format on files not previously processed and automatically on files freshly uploaded.  Thank you very much.

closed 2023-09-28 by jneilliii