jneilliii/OctoPrint-PrusaSlicerThumbnails #116: update to 1.0.3 failed

kb-, 2023-09-28:
I couldn't update from 1.0.2 to 1.0.3. There was an issue with Pillow 10 installation.

[plugin_softwareupdate_console.log](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/12754134/plugin_softwareupdate_console.log)
[octoprint.log](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/12754135/octoprint.log)
[octoprint-systeminfo-20230928234125.zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/12754146/octoprint-systeminfo-20230928234125.zip)

jneilliii, 2023-09-28:
Dang, I was patching to version 10 because of a security vulnerability and seems they dropped Python 3.7 support.

jneilliii, 2023-09-28:
I've updated the version pinning and re-released version 1.0.3. This should resolve your upgrade issue.

closed 2023-09-28 by jneilliii

kb-, 2023-09-29:
Thanks for the quick fix! It works.