jneilliii/OctoPrint-PrusaSlicerThumbnails #117: [FeaturRequest] Current Print Thumbnail a static URL

bdynamic, 2023-10-05:
First thanks a lot for the great Plugin! 

I would like to suggest a (rather simple) Feature: 
It would be great if the thumbnail of the current running print could be exposed via a stable url
e.g. http://<octoprint>/plugin/prusaslicerthumbnails/thumbnail/current_print.png
If no print is running some dummy picture (e.g. the Octoprint Logo with text "No Print running" could be displayed.

The purpose for this feature is that external SW, e.g. Homeassistant could show the current print image.

Thanks for considering
Birk

jneilliii, 2023-10-05:
I know others on the Discord that have made this work with HA as is, because the thumbnail URL is included in the file metadata. Let me poke them and see how they did that.

jneilliii, 2023-10-05:
Thanks to Sleepy2 on Discord, this is how they do it. 

> You need to have the Home Assistant Discovery plugin to get the print file name, then I just created a generic camera and used 

```
{{ "http://192.168.20.121/plugin/prusaslicerthumbnails/thumbnail/" + states("sensor.octoprint_print_file").split(".")[0] + ".png" }}
```

bdynamic, 2023-10-16:
Small hint for everybody using this solution: The following line is also working if your filename contains dots: 
{{ "http://<octoprint ip>/plugin/prusaslicerthumbnails/thumbnail/" + states("sensor.print_file").replace(".gcode",".png") }}

--> Replace <octoprint ip> with the correct IP adress

closed 2023-10-16 by jneilliii