jneilliii/OctoPrint-PrusaSlicerThumbnails #120: cached thumbnails in `data` folder bloat backups

lifeisafractal, 2024-01-21:
when running backups the cached thumbnails can make the backups rather large. As far as I can tell there isn't a good reason to back these up as they can always be regenerated from the gcode. If this is the case, it would be easy to exclude this plugin's whole data folder entry from being backed up with a simple [backup hook](https://docs.octoprint.org/en/master/bundledplugins/backup.html#additional_excludes_hook). I'm happy to open a PR for this, but wanted to see if this is a good course of action before that.

jneilliii, 2024-01-21:
I would be open to a PR for that feature as long as it's a configurable option that is disabled by default to maintain current behavior.

lifeisafractal, 2024-01-22:
Sounds great. I'll work on a this and open a PR against this issue when I get it going. Cheers 🍻

lifeisafractal, 2024-01-22:
Also, thank you so much for all your great work on OctoPrint plugins. It truly makes my 3d printing experience so much better!

lifeisafractal, 2024-01-24:
I implemented this [here](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/compare/master...lifeisafractal:OctoPrint-PrusaSlicerThumbnails:backup-exclude?expand=1), however after doing that I realized there is already and option in your plugin to use the uploads folder to cache the thumbnails. The backup system already has an option to ignore that folder, so it feels like it's not worth adding more code for something that is already achievable as is. I'm happy to open a PR if that makes sense to you, but otherwise we can close this issue out.

jneilliii, 2024-01-24:
That option is actually not what I recommend using for most people, it's only there for multiple instance OctoPrint sharing the same uploads folder like on a NAS. Please do open a PR because I think this option could be helpful for others.

PR #121 (2024-01-25): Exclude thumbnails from backup logic
Skip backing up cached thumbnails if 'uploads' is excluded from the backup.

This should make sense for any reasonable use case. If you aren't backing up your gcode uploads, then the thumbnails are useless anyway, and the user likely is trying to keep down the size of their backup archives. If you are backing up your gcode, the thumbnail is likely a good deal smaller than the gcode and the user likely isn't too worried about the size of their backup archive and this will give a faithful snapshot of the state of Octoprint at the time of backup in reguards to [... cut]
closed 2024-01-25 by jneilliii