jneilliii/OctoPrint-PrusaSlicerThumbnails #124: Doesn't find thumbnails

smurfix, 2024-03-28:
[image]

```
~/xl/uploads$ grep thumbnail test.gcode
; thumbnail_QOI begin 16x16 596
; thumbnail_QOI end
; thumbnail_QOI begin 313x173 40632
; thumbnail_QOI end
; thumbnail_QOI begin 440x240 70208
; thumbnail_QOI end
; thumbnail_QOI begin 480x240 76452
; thumbnail_QOI end
; thumbnail begin 640x480 105404
; thumbnail end
; thumbnails = 16x16/QOI, 313x173/QOI, 440x240/QOI, 480x240/QOI, 640x480/PNG
; thumbnails_format = PNG
```

smurfix, 2024-03-28:
Using octoprint 1.10.0rc3

jneilliii, 2024-03-28:
Please share actual gcode file for testing with directly. Your grep helps me none.

jneilliii, 2024-03-29:
Wondering if the mixed QOI/PNG formats are messing with the plugin. QOI support was added based on supplied gcode in https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/issues/114#issuecomment-1736469697. Looking at that file there is no PNG section and their settings section shows

```
; thumbnails = 16x16,220x124,200x240
; thumbnails_format = QOI
```

so maybe if you update that thumbnail setting and remove any references to PNG it will work?

smurfix, 2024-03-29:
Here's one of the files.
[test.zip](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/14801952/test.zip)

jneilliii, 2024-03-29:
I removed the metadata and deleted the thumbnail and ran the scan files process and like you is states no thumbnail and those are extracted and then displayed. If you don't have inline image setting enabled it will show up as a new button in the file list to open a pop-up. Maybe you just need to enable inline image in settings to show the thumbnail in the file list?

closed 2024-04-20 by github-actions[bot]