jneilliii/OctoPrint-PrusaSlicerThumbnails #125: Plugin fails to update thumbnail due to reliance on path/filename

theelous3, 2024-04-04:
Hi there.

The plugin fails to notice that a file by the same name is not the same file. For example, if the workflow is to iteratively re-export gcode to a file `example.gcode` with new gcode, the thumbnail remains unchanged even though the file is entirely different.

Looking at the documentation, there seems to be a useful event type `UpdatedFiles` which should indicate that a file was updated (I can see octoprint correctly recognises these same-name updates in its rendering of the `Uploaded: $human_readable_time` in the file list, so presumably this same hook will work.)

https://docs.octoprint.org/en/master/events/index.html#file-handling

Furthermore, via the `octoprint.filemanager` there is `"hash": "<sha1 hash>"` available to use to correctly check the underlying reality of whether or not a file is the same as it was, rather than something as weak as file name.

https://docs.octoprint.org/en/master/modules/filemanager.html#module-octoprint.filemanager.storage

theelous3, 2024-04-04:
That's under the add and remove event types- does that cover it? What is considered adding vs updating according octoprints lib? If saving over a file counts as updating it (which presumably it does) the linked line is never reached, as a file added which has a thumbnail by path isn't going to happen. Either way, I don't think that linked line is being hit.

> If you could expand on where the image doesn't actually update

It doesn't seem to update anywhere. All three of these images are from many prints ago. (Screenshot below).

I went and looked at my logs to check the timestamping. As suspected, all of the file timestamps are the same date `2024-01-31:18:38:37` which is essentially when I switched to octoprint and started using your plugin. Here are the selected logs from today only:

```
2024-04-03 15:33:12 - tornado.access - INFO - 200 GET /plugin/prusaslicerthumbnails/thumbnail/printme.png?20240131183837 (127.0.0.1) 1.00ms
2024-04-04 12:14:37 - tornado.access - INFO - 200 GET /plugin/prusaslicerthumbnails/thumbnail/printme.png?20240131183837 (127.0.0.1) 1.02ms
2024-04-04 18:43:30 - tornado.access - INFO - 200 GET /plugin/prusaslicerthumbnails/thumbnail/printme.png?20240131183837 (127.0.0.1) 22.01ms
```

Ah, maybe here is the problem:

https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/blob/3a6b7f126e48ef51f029d14ca0d1605b04fe9c61/octoprint_prusaslicerthumbnails/__init__.py#L282-L283

If the path exists, remove it. Then later, if the path exists, do update (but path is removed).

https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/blob/3a6b7f126e48ef51f029d14ca0d1605b04fe9c61/octoprint_prusaslicerthumbnails/__init__.py#L287-L288

I haven't looked at what error handling there is going on, but this should raise a `FileNotFoundError` if it's the case, but may be suppressed such that you didn't see it.

Screenshot:
[image]

jneilliii, 2024-04-04:
I'll have to go back and look at all this logic again. The issue was supposedly resolved in version 0.0.2...

> Thumbnail will now be overwritten on new uploads of same filename
Re-uploaded files will get url with timestamp to bypass browser cache

This might be some weird race condition though if the add event is fired quicker than the remove event is able to complete the deletion of the file or something like that, so very well could be in those conditionals. 

> That's under the add and remove event types- does that cover it?

Yes it should. I don't think that logic has been changed and the UpdatedFiles event is not what you think it is, it's related to the whole of the file list I believe and not an individual file being overwritten. 

Quick question, if you look at the actual files in ~/.octoprint/data/prusaslicerthumbnails/ are they getting updated and the metadata is just not getting the new timestamp?

jneilliii, 2024-04-04:
> If the path exists, remove it. Then later, if the path exists, do update (but path is removed).

not exactly. the file is extracted just before this line, so if the extraction doesn't complete it might not update properly. 

https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/blob/3a6b7f126e48ef51f029d14ca0d1605b04fe9c61/octoprint_prusaslicerthumbnails/__init__.py#L286

theelous3, 2024-04-05:
> Quick question, if you look at the actual files in [...]

No they are not updating.

> if the add event is fired quicker than the remove event

Shouldn't be the case. Events are blocking in the context of the event manager:

> Please note that until your plugin returns from that method, further event processing within OctoPrint will block - the event queue itself is run asynchronously from the rest of OctoPrint, but the processing of the events within the queue itself happens consecutively.

from https://docs.octoprint.org/en/master/plugins/mixins.html#eventhandlerplugin

However, looking further, I see you are manually calling `on_event` in a thread you start from an api call hook, which also deals with `FileAdded` events after a file list scan: 

The `Thread`, follow this:
https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/blob/3a6b7f126e48ef51f029d14ca0d1605b04fe9c61/octoprint_prusaslicerthumbnails/__init__.py#L360

The call:
https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/blob/3a6b7f126e48ef51f029d14ca0d1605b04fe9c61/octoprint_prusaslicerthumbnails/__init__.py#L303

I guess this will break that blocking guarantee. Perhaps a simple lock acquired by your implementation of `on_event` would solve it. Forcing either your hook's call or octoprint's call to wait for the other. I'm guessing it's probably best to not manually call `on_event` but if that's the issue, a lock should be an easy way out.

jneilliii, 2024-04-05:
> However, looking further, I see you are manually calling `on_event` in a thread you start from an api call hook, which also deals with `FileAdded` events after a file list scan:

This could be the case, but only if you have the option to rescan the files when refreshing the file list. Could you share a screenshot of your settings for the plugin? I'll be back from a work trip tonight and might be able to debug this further over the weekend.

theelous3, 2024-04-05:
It is on. If this isn't on by default, it's on because I was messing with the settings trying to fix the issue. 

[image]

jneilliii, 2024-04-05:
Ok just got home and couldn't reproduce this issue. There may be a plugin conflict with something you have installed that I don't maybe? 

https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/assets/5249455/4d59763e-5011-40e5-ab09-fe65d150fd4b

theelous3, 2024-04-05:
Ah. Haha. So, I'm not uploading. I'm saving the file directly over the other file via the os, and slapping the refresh button.

theelous3, 2024-04-05:
Sorry I just realised that is some vital information. I don't use octoprint like most others (it's local) and completely forgot other people are uploading files.

jneilliii, 2024-04-05:
Ok, yeah. That makes way more sense now. That's not supported officially by OctoPrint, so any "backdoor" method of over writing files doesn't go through the event queue at all typically. The best approach, which is officially supported is using the "watched" folder and that will be processed by the same event queues as the upload process does.

theelous3, 2024-04-25:
Since we last spoke, the issue has spontaneously dissapeared. I am presuming an automatically applied update on octoprint side. Thanks for the activity, engagement, and plugin :)

closed 2024-04-25 by theelous3