jneilliii/OctoPrint-PrusaSlicerThumbnails #126: When uploading and printing files from Orca, plug in is not registering JPG.

Cdewitt70, 2024-04-21:
Using Orca 2.0.0, when I upload and print from Orca it is not showing the JPG or giving the image icon on the file. However, if I reboot my Pi, then upload again, the first file will show and then it goes back to not registering. 

I have verified the slicer is exporting with the JPG info at start of gcode. Image and name of file both show up on my printer's display when printing from SD but do not when printing from Octo. 

Can share logs if needed. My settings seem to be correct from everything I have checked.

jneilliii, 2024-04-21:
Please share exported gcode file from Orca that was uploaded.

Cdewitt70, 2024-04-22:
https://drive.google.com/file/d/1BUwDaPLuUQJEvDpkyH0xViBWHCECVagW/view?usp=drive_link  

Gcode couldn't be directly uploaded so I did it through google.

jneilliii, 2024-04-22:
you can change the extension to txt and you can upload. Link is giving me prompts to request access.

jneilliii, 2024-04-22:
or compress it into a zip file.

Cdewitt70, 2024-04-22:
[WD-40_Big_Blast_Replacement_Nozzle_PETG_23m35.txt](https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/files/15057827/WD-40_Big_Blast_Replacement_Nozzle_PETG_23m35.txt)

jneilliii, 2024-04-23:
So I was able to take a look and it appears that Orca changed their output format for whatever reason so the regex doesn't match the first line indicating the start of the thumbnail block. 

```
; thumbnail_JPG begin 250x250 4293 1 77 3 7 0.24 1.75 1.27 67
```

the bit `1 77 3 7 0.24 1.75 1.27 67` at the end is breaking the plugin's ability to extract the image data.

Will have to look into why they did this to determine if I'm going to tweak the plugin or report as an issue on the OrcaSlicer repo.

jneilliii, 2024-04-23:
So I just tried 2.0.0 with my printer profile and it didn't add that extra information to the file. This may be printer profile specific. Can you share your printer information and a screenshot of the thumbnail settings.

Cdewitt70, 2024-04-23:
Yeah I am using an Ender 3 S1 Plus. 

Here is my thumbnail settings

[image]

jneilliii, 2024-04-23:
Sorry, I meant the settings in OrcaSlicer not the plugin.

Cdewitt70, 2024-04-23:
Which settings do you need? Sorry I am ignorant to this type of thing.

Cdewitt70, 2024-04-23:
"C:\Users\Chase\AppData\Local\Programs\Python\Python310\python.exe" "C:\Users\Chase\AppData\Roaming\OrcaSlicer\user\default\process\E3S1PROFORKBYTT_printdata_orcaslicer_v19_thumbnail.py";

this is my postprocessing script, due to my printer I am using Thomas Toka's firmware for the screen preview. And his fork of your plugin had this. As well as this for Layer change G-code 

;AFTER_LAYER_CHANGE
;[layer_z]

jneilliii, 2024-04-23:
That's why I couldn't reproduce with any of the standard thumbnail options. I recommend you post issue on his fork.

closed 2024-04-23 by jneilliii