jneilliii/OctoPrint-PrusaSlicerThumbnails #127: Thumbnails not being created automatically

LariOesch, 2024-05-05:
After updating OctoPrint to 1.10.0 thumbnails aren't being generated automatically.
Thumbnails appear after scanning for them via Settings > Slicer Thumbnails > Scan files, and returns Scan Results _no_thumbnail_ after which thumbnails are displayed correctly. 
Enabling the "Rescan all files when pressing refresh in file list." -setting works, but itäs still not automatic, like it was before.

Octoprint 1.10.0
Slicer Thumbnails 1.0.7

After uploading:
[image]

Only after refresh
[image]

No logs that mention anything related to plugin.

jneilliii, 2024-05-05:
The extraction logic is no different between using scan files and upload. Enable debug logging for the plugin in OctoPrint's logging section of settings and try again. Afterward share octoprint.log.

jneilliii, 2024-05-05:
did a quick test to validate and here it is working as expected in 1.10.0. 

https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails/assets/5249455/b9da57a7-d85e-44f4-b63d-38f2ea72ce26

LariOesch, 2024-05-05:
Odd. Maybe the Octoprint and python update process threw something off. I tried re-installing the plugin but that didn't fix it either, but creating new Octoprint instance and installing plugins there does indeed work as expected, so there's probably something else that breaks the functionality.

I'll keep on investigating, but I believe this isn't plugin issue after all and this can be closed. 
Sorry for the inconvenience 😅