jneilliii/OctoPrint-PrusaSlicerThumbnails #128: Image button is not there

alanglois97, 2024-06-19:
Hello,

I use Octoprint and everything is up to date. The thumbnail slicer is installed but i dont have the image button showing... Is it still available? If not, any other options?

Thanks alot

[image]

jneilliii, 2024-06-19:
it looks like you have the inline thumbnail option enabled. that makes the button do away.

jneilliii, 2024-06-19:
based on the broken images shown in the inline option, it seems that the files are missing from where expected. please share screenshot of your settings and upload one of your gcode files here for further investigation. what slicer do you use to generate gcode, etc.?

alanglois97, 2024-06-19:
Hello, thank you for the quick help!

I am using prusa slicer

Here is a screenshot of my octoprint settings, i disabled the in line option
[image]

Here is another one of my prusa slicer settings, Vignette means thumbnail
[image]

Yes based on the broken picture icons I feel like it might be the export of the thumbnail from prusa that is not ok but i dont know whats wrong

Thank you for your help
[test thumbnail.zip](https://github.com/user-attachments/files/15902960/test.thumbnail.zip)

alanglois97, 2024-06-19:
Working, thank you again

closed 2024-06-19 by alanglois97