jneilliii/OctoPrint-PrusaSlicerThumbnails #131: Thumbnails broken when creating one from Prusaslicer 2.9.1

magiwogg, 2025-04-02:
Hi. it seems that my prusaslicer or the plugins seems to showing broken thumbnails.
I set all the settings as following, as explained in plugin page and repository. Am I doing something wrong?

[image]
[image]
[image]

magiwogg, 2025-04-02:
here is the beggining of the g-code for your reference, which seems to creating thumbnails
; generated by PrusaSlicer 2.9.1 on 2025-04-02 at 00:21:00 UTC

;
; thumbnail begin 16x16 824
; iVBORw0KGgoAAAANSUhEUgAAABAAAAAQCAYAAAAf8/9hAAACMUlEQVR4AYWTT28SQRjG+TZ+BFOaAs
; sS7aXR1J6NZ+PBGDUkjYQYIglKIIvYP4IGkdDdlrW7LGwXEFaWQmmlBbE2HIgmejLGEEPCoYfXfTed
; Bgi0p5mdfZ7fPPPOO6ZQKPQ5EolURVFUPB7PcSwW01KpVMHpdHZ5ns/H4/Gy2+0+URRFikajVVyXZT
; mDvnA4vG8SBEG5TDQNnkgkVBOKAoHA4STRMByBCHe5XB2cB4PBBvpMk0QkDREReDKZVIfhRgKfz9di
; GKaBIpyPixBOdiXwkQTpdHpnXLTO3OxvvVv4x79f7HGxW39z/GJvO3GjzybXahcmeLP+uCuxSz1l0w
; zZjRmQOTMU+FljLH6YhVxqfhBZufsLNyK+8wRI1ZR7P8vpOchtmWH5wRUoCRbQMlbIsjNQU6xQyVpA
; FenTZ56H39E3kkASE+quLqjnrCDEr8JqYH6A342iHZoqbazv6ZD9vBXk9Kv6eQJSKObF7T8Yt/WJNo
; S1HatuovSRgq8VBxypFNTPAIz/zu+JNdAyjtNG0QYHBRscFik4yNugrV2DL2UaKpk5ON6ldaAFBP51
; FX0jfYA1KClPv2HsoxKlG+3QKtsN80nVYRylrdEgbD7p4FVO7YMU97y5l1/qN1U7VPVjdOrX9RpQUJ
; EXBhvx5Y7X622P3MJwAkmSZJyjSNx+W14L3/+x+vJRl02u1NDAsmyR47iPFyYgIoSQ1zgOH+lE8oOI
; sBOHRX6/vzkNjgn+A3TboZiYCsuEAAAAAElFTkSuQmCC
; thumbnail end
;

;
; thumbnail begin 100x100 11664
; iVBORw0KGgoAAAANSUhEUgAAAGQAAABkCAYAAABw4pVUAAAh8klEQVR4Ac2d158VVbbH6/+4k53PjO
; OMGQmdM6HBgI4OOmMYYRRzQEUd0Qs4iiNBgmQESZIzTWxylIYmo8iod+bejw8+8ODnwwMP++7v2rVO
; 7y6qzqmq7hseNpw+p8Kq9Vvht9beVRU8/fTT/1q4cOFe/q+pqTFx49577/1h0qRJbR988MHJgQMHXo
; n+3tTUdHXs2LFn2Sbudx2cY/r06UeHDRv2XdzvfJ9GFs7VXbIknUtlefHFF79JOgbHLyXLqFGjvkAW
; 5E46DudSWYJSgvsHHzly5CV2fOSRR77X74cMGXJ57ty5B4sJHgcuF8Ix9fs0gsfJ4oOrsvBbFllQaF
; dk4dqjsrAv32WRBTmCNBv7g4vWEzHSCp6kULXUtILHyYIC2R8w+C6vLM8999w/uyILuvBl8Q03rSxB
; 1hPryVtaWjbh0lnB8Md77713ev369Vuj3pJVlo0bN7YgS1YworJwnK7Ksnr16u2LFy/enRUMHUGacO
; UP3BP0cU8/9mW5iGhO0jicFVzNf8ih3tJVWbKEq2KycIyssoiHFEts/kBYThLNOaWSbJLg0XOWSrJR
; wTlfVBZN6F2VxTe0NLLEkYissiiJCJISW9zGSQImJdk0ghez1mKyFCMRpdhcGvaTxtDSEJo0hubLEk
; QVGo19WVzYT2w+uFnYT3fJksTmsrCfYoaWhdAkGX2cLEFUobqB/7nUCeMuAgFQaF72052yKLhdlQWD
; 6C5ZNBdHjS6I25GQsGrVqh1pa4u4gYtu3bp1IwLkZS3st2DBgn3dIQussKuyzJkz5+CaNWu25aHFcb
; LEhcLA/8N3rSxJNu6kyjiyJLY0smRVaJwseeomJRFYdFdl4f+k8BvEbewrJYtCk9hPVnC7S5Y4EpFV
; ljhCk5VBJckSVzYEpdhPsSQbJ3hSaEnD5kqxnzRsjlGK/aRhc4xSJCINuKUITdTQgrR9KD+ZRRWalv
; 0UAzcL+0lic4y07Ke7ZClmaGlJhC9LkCWe+gyK/bI20JLAzcN+omyuu2RJYj9pZNH9FCQMJqssQZYd
; /B3p2WgLJc8xuIhx48adgrUgfJ5jqCzLli1rRZY8BMSXhb4asnSll4Us9LLyyhJk3UHRHz9+fHteBs
; XQxP3GG29czMvmUByyTJky5Vh3y5IVFCU0yJK3HyaAFEuySYL7yivVpogTPEoisjIoRhz7ycqg4mQp
; NelUTBY/F2fph6ksABmkYVA++4kTMqlNER3F2E9aBsUoZoFp2JwvS1zOSZp0yiMLcpQC189/gQqXxK
; DSJstS9Dgt+/HbFP+TsqQhEcXYXHfLor8FcV6gSsvDfvQisDCOmYf9+AxKz52H/UQNTWXJwn6ibK47
; ZdGo4m8XRHfERefNm3eAvk104ywXwb4I3RUmxkWoLFlyXVQWAPjkk0/2awslrywzZsw4gjxdkQVAis
; kSRL8gCS1fvnwnLCova9HJLITPy6BUFigkzCWvLMp+Zs2adej/gywYKsAmsbkgurEm7ixJ1h9R9pO2
; TREnuE8i8jQ7o+wnD5uLC+V5ZIkSmiQ2F/APsTCJ/RRLbNGRxDjS9sN8weNyTlo2l1aWUoZWLP+lZX
; OMJEITx+aCNOwnLrGlFTyq7CQ2p4KXIhGlwM0qS5Kh/W/LokYfZGE/cSs7orSt1IgLAXnYT5TN8V1W
; 9hPH5vL0oeIMDR1lITQqS5D2pP6OhIzJkye3pXXZuKGV7Ntvv30hL/vpLllUoaNHjz7XFVkAUZN+sc
; UcxUaQdQdFn0VlXW3oocilS5fuziu8ykJTsKuysFCuq7IAJLJAa/P0wwSQrG1mn/1kSWxRwZX9ZGlT
; xFmkPxWQlc0xouwn79R1VJY8zU5pvxdLssUE9xWTlkFFBY8ePy2bS0qWWWVJIjRZDC2NLGkMTXNxEJ
; 10KrZxMSZWikGlYRyl2JwvSzH2052yFCMJaUhEKTYXJRFBdEc/9mVlP3EMKq3gOvo2VZsxo35/+dPZ
; f/nX2L+WmbdHlpu3Xi0zzz5RbiZPeOXS5El//eKuQdVm6COV5q+vVJqPPywz0+yY9F65+fuYMvPq8+
; XmvsFVprm5sSDLg0Oarzw/vMoeq8rM+qju6vplf/5u4ew/fv/UsEpz/73VReWJAzdrqE4yNH9FjH4X
; RHfU2MfS/LyMQxnUq6++eimt4C88VWHmTyszm5f1MVtXlJkW+//mz9zYsLi3Wbeot1k5v6fZs77CrP
; m0l/lkym1m8cweZsXcO+T7Zfb/jUvYrpdZ+UlPs2hGTwtkuVk+f9APW5eXme2rys3ejZVmnx0cY9fa
; ctO6ptxsW1lmj9PHjHypwgxsrippaC+88MK3pW4qKgUuBprExIK4k8MS6Gd1ZXEaF6BMrFRimznRKm
; dthYydVkmMHautsgDGArJuYS87LCDzehaUynejR/7GzJl0q5ltBwAx+HvVgp4CKMAytizvI4rfaY/J
; 5wObq8z+TRaYDRVmN+CsqxDAVi1wwDQ1VceCQj+MRXvdsVAuSS9B1LKV/WRpUyRZE8coldimj3cK3r
; +pSpS9Z0OlKAhQAAiFAsRq6wWfTrvdTH7vJgeKVegmC9bEsTcKAJPevdF6xe1m1oRbBKDPZvcwq+w+
; jDX2901LewtI6xf3EnD2hcDu3YDHuHNiBACzfF5v89ifKjtZtrZz0k46JRmp6iWJzQVxG/vKzcJakp
; JlUmK7+84ac7ClSoZY7WYHyraV5RJ+NtixdPYdEoIYH//9ZgvKbXYbtz37rV7QS76fP/U2s/Dj2wWA
; ZXPuMJ/Zwf+EM8KY/r2CY4XfAQgesntdRQEQB0qZPW4f8+TQilgS0V16ictFQRrGEdemiI5S7CeuTU
; FSPry1WgbKJbZLeLL5gv9RNqCQR1A0HkDeQHkAwlhvt5098VYBBNDYlv/FM2yuWT7vDgtqD+shfcyn
; FrD1Nh9ttN6y1h6f7/AaCWdrygsewt945vol1Vf/NnZEUb2UKhtKEZoomwvSsh8/4ftelLX38/DDf7
; g8d87HR58aPvS7caPrrx7d3gEI8b3FJmBCEWAAylqr1FVWwStQ7CwXhlAYnoQSJ4z5nVk+1/3GPoCG
; ogESQCAJeBtA4CGADWgovCOvlMk+G5f0kc/sCzgty8rM5Pf7CPMrppe4siErE1Nwg6zlvR/74mhbdB
; CL37HUdc6kPjaWOwaE4ggdKP/ItmoZWPvG0BtQGorV7WFZK+Y5q0dZhJrWNRVm6ribJVwdtfvjJYQe
; FL0mBHNTeKz1dp8t9nv1HvW6baucZ0jIsvkKMNhXCQF/M0a/UWHpcVVRvfiLsJPuzEpj9EGWHXw09a
; bPpBAFv184o49YGVbK2BTSWKxx8zJniVgpfxN6UDgegTLZjrCCchhsw8ByAeSDd35rpn94sygT7wIw
; EvfRbTVyLhRNLmA4OtxblIuiySd4GMfiHIRAEjv5RPMILOyQ5LZKc2CTS/5zJlcYapckveAVK1eu3N
; Glmz6zeohOZo0YMeLrpMQ29NEqUXRHkiwvfJZQsNxZ7salLpzgAQBA2GE/VQzbAALfcQyAIc8oGNtt
; 8ufvw1trLDurEkDZ99AW93lvyKIY7Itnch6osSjaeiXAYRiAIgl+vaPfgM5xGGyHF7t8Z4vLSeVmUH
; P1NWCgCxbbZV2r1gmQYm2K6IhjYtHE1rdvjdmxyoWBPeEF8ln+Rjn2QrFSFO6suCORAhCxHiUABtvt
; tKGJ8MS2H/3tRjPB0lxAxIJb17oir313vTm1t1EsH3BRHsdfbenu/jD5o2DyxWezXZJH6Wx30J6r1R
[... 82 lines cut ...]
; 3wvOMudXuzPLMjjollfQuAf2fWkPtrr378YaXEcgk/4aSQVur8r81DPMh1fcsl8dLhvdTWLMzIhSw3
; XMhx+yt91WMACt7Fvegci4V4VN+6xnjsm70MQGR9GlwcoclaNhTmQ9I8Xq9UCyXt4/WSbuO6a1CZef
; NlnsRg593XVkvX9thOEmqD1AEsnJOJoHCCyN0MVB+u03IMTJPxl14toUleCz5aNJqz/DFzQvk1oSkN
; syxFf9OWDf5kVqBfJN3Cm7aFUuyBLGl5+8AB5eapYWVm7tQ7f2hZeecP5JlrFqaFM3RuTrzOrdk95C
; jxRZuc8bBLxwZITqAvdmpfo9zHDllA+ey3ekHl1Y/HN14ZPrRX4XaHYmEobpFglsRd7L71a+bUowr1
[... cut ...]
; SGrnsF+Wx+slkYgshlbsnVlZDK3wJIdSrKWY4FHBooktepGlJrO6S5ZomyLu97RvRygmS5q3RpQytK
; gsga9Qdoy+BSAL44hjUL7gadlP0hsJulOWtCSimCxZ3hoRB24cEwuiO3IS0OSpo3nZj16Evh0hD/vx
; izuVJQ/7UUPL8qaGYuDq2xHyyoJexowZc67k87JqPBfihF19OwL7pn07QtJgP5UlzzooX5ZibyRICw
; qUVt+O0BVZVC8lnyjns5+0bYo4wf3EnYdBMXwSkZfNdbcs+hiSvLL47ZwkNhfECe4fpFRi8wfbxK1B
; ysKgkiazsrA5RlI7J4ssjCQSkYXNFXs7QhTcoBht8y8umtjSCh5VaLG8lPZNDcXYHKMU+0ljaGlkKc
; XmfFlKvTVCZflvN7h93W3LSnwAAAAASUVORK5CYII=
; thumbnail end
;
; 

; external perimeters extrusion width = 0.42mm
...

jneilliii, 2025-04-02:
I'm not quite sure why it wouldn't be working, the comment tag looks right for extraction and it seems to think the file extracted since it updated the metadata with the URL to show in file list  do you see the file in the plugin's data folder at the os level or see any errors in octoprint.log?

magiwogg, 2025-04-02:
I tried to dig deep in with ssh and look into file, but before I could come up with long-forgotten octopi password for ssh my octoprint suddenly showed the thumbnail. how strange!
rebooting octopi did not worked so... maybe power off -> on method during attempting root login worked?

[image]

closed 2025-04-02 by magiwogg